kuznia-rdzeni/coreblocks #237: Migrate to Data type

lekcyjna123, 2023-03-06:
RFC with Data types have been merged, so we can migrate on it.
https://github.com/amaranth-lang/rfcs/pull/1

tilk, 2023-03-06:
Removed the "Priority" label and swapped for "Nice to have". We kind of settled on records, and work around most of their limitations.  Amaranth's data types functionality is fresh, bugs are to be expected, the limitations are unknown, also the functionality is evolving (see e.g. https://github.com/amaranth-lang/rfcs/pull/8 and https://github.com/amaranth-lang/rfcs/pull/11). I'd wait a little before attempting to switch.

whitequark, 2023-03-12:
> I'd wait a little before attempting to switch.

Agreed; I would suggest you wait before the next release.

whitequark, 2023-05-18:
> We kind of settled on records

Note that Records will be removed with a 1-version deprecation cycle (so probably before the end of this year).

tilk, 2023-05-18:
> Note that Records will be removed with a 1-version deprecation cycle (so probably before the end of this year).

Thanks for the heads-up. Do you recommend that we still wait for the next release to work on an alternative solution (using amaranth.lib.data), or is it better to start ASAP?

whitequark, 2023-05-18:
`amaranth.lib.data` should be in its final form now. I would recommend starting to work on it ASAP, and any bugs or issues you find as you start using it are most welcome as reports since we have the most flexibility to amend it before it appears in a release.

whitequark, 2023-05-18:
Also, nothing technically stops you from copying `amaranth.hdl.rec` to your own codebase and continuing to use it, as it does not integrate with core Amaranth directly, only using public APIs.

tilk, 2023-05-18:
> Also, nothing technically stops you from copying `amaranth.hdl.rec` to your own codebase and continuing to use it, as it does not integrate with core Amaranth directly, only using public APIs. It's just very poorly designed.

Yes, yes it is. We did some lot to work around the deficiencies. For example see this function, which allows to assign fields, with error checking, between records and dicts with signals, and more: https://github.com/kuznia-rdzeni/coreblocks/blob/739b1235c73230725fece3d27ad05c0203da942b/coreblocks/utils/utils.py#L146

PR #357 (2024-02-03): Migration to amaranth.lib.data
As Records are basically deprecated, I've refactored the code to use amaranth.lib.data instead. Most of the changes are pretty straightforward. A weird one was RS, as it mixed combinational and sequential assignments to the same record. I still left records in Wishbone code.

Closes #237.

~Edit: damn it, the debug signals. I wonder if this can be easily fixed, or maybe I just found a major issue with `amaranth.lib.data`, to be reported?~

~Edit: this looks bad (https://github.com/amaranth-lang/amaranth/issues/790), the migration should probably be postponed.~

G [... cut]
closed 2024-02-03 by tilk