kuznia-rdzeni/coreblocks #289: Generate compiler options from `isa_str`

piotro888, 2023-04-15:
We use some of the extensions (like Zmmul) that for some reason are not supported by compliers. We should generate superset of our internal isa_str that is accepted by gcc or binutills as `isa` valid paramter. This is needed in core tests generated from assembly.

Example `rv32i_zmmul` -> would be converted to `rv32im` for gcc and binutils (possibly with extra parameter `-no-div` for gcc).
Disscussion for Zmmul in gcc: https://groups.google.com/a/groups.riscv.org/g/isa-dev/c/M-37xaIllsw, change was not accepted in 2021 because this extension was not yet frozen, but now it is ratified since 2022 (https://github.com/riscv/riscv-isa-manual/blob/master/src/m.tex, maybe we should file an issue to gcc?). Anyway, problem would probably exist for some other extensions too.

piotro888, 2023-04-16:
For Zmmul, it turns out that support for it was commited 7/8 months ago to gcc and binutils; and it should be available in the latest release of binutils 2.40, but not yet in gcc (we have to wait for next release).

tilk, 2023-04-18:
I've backported binutils 2.40 for Debian stable, and I can confirm that it understands Zmmul.