kuznia-rdzeni/coreblocks #301: TestICache fail with round robin scheduler

lekcyjna123, 2023-04-23:
When we switch scheduler in transaction core from eager deterministic to trivial round robin (I have done that by changing default in definition of TransactionManager) then test `test_1_way (test.frontend.test_icache.TestICache_0_blk_size16B_rv32i)` stuck in infinity loop.

lekcyjna123, 2023-04-23:
@xThaid Can this issue be connected with the problems seen in Icarus?

xThaid, 2023-04-25:
> @xThaid Can this issue be connected with the problems seen in Icarus?

It is not related. I will take a look at it anyway

xThaid, 2023-04-28:
Round robin scheduler creates a combinational loop (logs from verilator):
```
%Warning-UNOPTFLAT: core.v:34414:8: Signal unoptimizable: Circular combinational logic: 'top.c_grant$221'
34414 |   wire \c_grant$221 ;
      |        ^~~~~~~~~~~~
                    core.v:34414:8:      Example path: top.c_grant$221
                    core.v:10338:3:      Example path: ALWAYS
                    core.v:33977:14:      Example path: top.c_data_in__exec_fn__op_type$276
                    core.v:18098:3:      Example path: ALWAYS
                    core.v:17978:13:      Example path: top.c.scheduler.rs_selector.forwarder.read_value__exec_fn__op_type
                    core.v:9991:33:      Example path: ASSIGNW
                    core.v:8288:14:      Example path: top.U$$0.U$$10_requests
                    core.v:11287:3:      Example path: ALWAYS
                    core.v:8287:14:      Example path: top.U$$0.U$$10_grant
                    core.v:11023:22:      Example path: ASSIGNW
                    core.v:34414:8:      Example path: top.c_grant$221
%Warning-UNOPTFLAT: core.v:34435:8: Signal unoptimizable: Circular combinational logic: 'top.c_grant$57'
34435 |   wire \c_grant$57 ;
      |        ^~~~~~~~~~~
                    core.v:34435:8:      Example path: top.c_grant$57
                    core.v:20261:3:      Example path: ALWAYS
                    core.v:34394:8:      Example path: top.c_enable_sig$58
                    core.v:10930:19:      Example path: ASSIGNW
                    core.v:34636:8:      Example path: top.c_run$78
                    core.v:9965:32:      Example path: ASSIGNW
                    core.v:8315:14:      Example path: top.U$$0.U$$2_requests
                    core.v:11789:3:      Example path: ALWAYS
                    core.v:8314:14:      Example path: top.U$$0.U$$2_grant
                    core.v:11075:21:      Example path: ASSIGNW
                    core.v:34435:8:      Example path: top.c_grant$57
%Warning-UNOPTFLAT: core.v:8297:14: Signal unoptimizable: Circular combinational logic: 'top.U$$0.U$$13_requests'
 8297 |   wire [4:0] \U$$13_requests ;
      |              ^~~~~~~~~~~~~~~
                    core.v:8297:14:      Example path: top.U$$0.U$$13_requests
                    core.v:11426:3:      Example path: ALWAYS
                    core.v:8296:14:      Example path: top.U$$0.U$$13_grant
                    core.v:10980:22:      Example path: ASSIGNW
                    core.v:34422:8:      Example path: top.c_grant$433
                    core.v:448:3:      Example path: ALWAYS
                    core.v:34360:8:      Example path: top.c_enable_sig$434
                    core.v:10816:20:      Example path: ASSIGNW
                    core.v:34603:8:      Example path: top.c_run$426
                    core.v:10006:33:      Example path: ASSIGNW
                    core.v:8297:14:      Example path: top.U$$0.U$$13_requests
```

And smart Amaranth instead of detecting the loop, tries to converge the signals and loops forever.

And it looks like it is not a problem related to the cache itself.