kuznia-rdzeni/coreblocks #384: Python 3.11 issues with VCD dumps

Kristopher38, 2023-06-07:
In python 3.11 some signals don't appear in VCD output (one clue: perhaps this happens when they're not a class member) or have width truncated to 1 (!). Needs to be investigated.

lekcyjna123, 2023-07-16:
Not sure if this is correlated but I have observed that giving a name explicit cause that signals appears normally in VCD.

lekcyjna123, 2023-07-19:
Reproduction:
commit: `07f0907f84c2715aa5f2f13a3aaadb8be24c66d5` on branch: `lekcyjna/vector-permutator-2`
Command: `./scripts/run_tests.py -t test.structs_common.test_scoreboard.TestScoreboard.test_random_1`

https://github.com/kuznia-rdzeni/coreblocks/blob/07f0907f84c2715aa5f2f13a3aaadb8be24c66d5/coreblocks/structs_common/scoreboard.py#L50-L51

Is signals have a name set as an argument, then both appears in VCD in `dut`. And their first non zero values are `0x11A` for `data_forward_valid` and `0x112` for `data_forward`. If `name` arguments from this signal are commented out:
https://github.com/kuznia-rdzeni/coreblocks/blob/8b3536e491c9fbb57c572b53db3a9084bb4259c1/coreblocks/structs_common/scoreboard.py#L50-L51
then in VCD in the same place are signals `data` and `data_forward` (no `data_forward_valid` and `data` was not present when these two signals had its own names). Values in first non zero cycle: `data_forward`: `0x11A`, `data` : `0x112`.

This suggest in my opinion that there are problems with automatic signal name recognitions in amaranth on python 3.11.  Real `data` signal from scoreboard is set synchronously, so it can not have the same value in the same cycle as `data_forward` signal which is set combinational.

tilk, 2023-07-19:
A minimized example (a short `Elaboratable` without dependencies on Coreblocks code) would allow to report the issue to Amaranth.

(By the way, with the recent release of PyPy with 3.10 support, I tried running Coreblocks tests using PyPy. They ran significantly longer. This probably means that we should stay on CPython and go ahead with the 3.11 upgrade soon.)

Kristopher38, 2023-10-10:
@lekcyjna123 thanks for the repro steps, I was able to replicate the issue and produce a minimal example, but later discovered that it already got fixed in (most likely) [this commit](https://github.com/amaranth-lang/amaranth/commit/c343e879d3e30c0f071345154b9b7176c9c63fc4) 

For completeness, here's the minimal example:
```python
from amaranth import *

class Example(Elaboratable):
    def __init__(self):
        self.foo = 1

    def elaborate(self, platform):
        bar = Signal()
        def _():
            m.d.comb += bar.eq(self.foo)
        self.foo = bar.name

b = Example()
b.elaborate(None)
print(b.foo)
```
Should print `self` instead of `bar` on python 3.11, on amaranth version before the aforementioned commit.

This was indeed a problem with automatic variable name recognition - it caused some signals to get their names incorrectly deduced - in the right circumstances they could take the name of other (local?) variables. Because of this I'm 90% sure that the issue with bit widths getting truncated to 1 was one of the observable effects of this - the signal in vcd dump wasn't named correspondingly with the signal in amaranth, so it could appear as if its bit width was truncated to 1 (whereas in reality it was an entirely different signal).

I'm closing this issue as we already updated amaranth to include that fix.

closed 2023-10-10 by Kristopher38