kuznia-rdzeni/coreblocks #403: Transactron verbosity and reporting

tilk, 2023-06-30:
Transactron should be able to report some metrics related to size/performance (number of transactions/methods, number of connected components, connection count, etc.).

lekcyjna123, 2023-07-07:
It would be nice to have also an possibility to check why a method is not ready.

On the beginning it was easy to check: if `ready` signal is up, then method is ready, if not, then not. Now we have a simultaneous transactions and methods which calls methods. Debugging why such complicated circuit is not running is hard. Someone has to know, what is behind abstraction level and it has to dig sometimes three or more layers of called methods/transactions to find which one is buggy and cause the whole execution to stop.

tilk, 2023-07-08:
> why a method is not ready.

Did you mean "not run"? Readiness is a signal produced in the method's definition and has nothing to do with the things you wrote.

Method is run when one of the transactions which calls the method is granted. A transaction is granted when its request is high, all of the called methods are ready, and a conflict does not block it from being granted. That's it.

If you want simpler access to the relevant information, I think that currently the simplest way would be to do this via exposing the data in the `TransactionManager`'s debug signals.

And by the way, that's unrelated to this issue, it should go to a new one.

lekcyjna123, 2023-07-08:
> Method is run when one of the transactions which calls the method is granted. A transaction is granted when its request is high, all of the called methods are ready, and a conflict does not block it from being granted. That's it.

Yes, this is understandable, but there is a problem with "all of the called methods are ready", since we have simultaneity and method nesting, finding a set of called methods isn't an easy task an a recursion can go by three or more layers of method each of them can call set of other methods.

> If you want simpler access to the relevant information, I think that currently the simplest way would be to do this via exposing the data in the TransactionManager's debug signals.

Yes, that is probably what we need.

tilk, 2024-03-22:
This was done in #535.

closed 2024-03-22 by tilk