kuznia-rdzeni/coreblocks #404: Synthesis benchmark for submodules

lekcyjna123, 2023-06-30:
It would be nice to have a benchmarks results for each part (submodule/connected component) which run in one cycle, so that we will know which modules are the biggest and most worth to optimise/split.

tilk, 2023-07-03:
As this is optimization-related, I don't think this is a priority; also, this would probably increase our resource usage a lot if run in CI.

lekcyjna123, 2023-07-03:
> this would probably increase our resource usage a lot if run in CI.

This shouldn't be run in my opinion in CI, because as you have said resource usage will be too big. It can be useful to have such script for local execution to summarise coreblocks state and get input for optimisations.