kuznia-rdzeni/coreblocks #410: Add script for checking mathod usage

lekcyjna123, 2023-07-01:
We use methods everywhere and expect that some methods will be called every cycle (or can be called every cycle). As for now we don't have tools to check if our assumptions are mirrored in reality. So please add a script which as an input will take a name of method in a module and as an output will report its usage over a time of running test.

tilk, 2023-07-03:
One can currently test this manually using VCDs, or automatically by adding a synchronous process to a test which checks this. Could you elaborate?

lekcyjna123, 2023-07-03:
Such script can be useful for optimisation. We can take a benchmark, run it and generate a VCD file. Next such file can be used as input for our script which will return us a usage of each method over time, and such data can indicate bottlenecks:
- methods which are supposed to be run frequently and they aren't
- methods which shouldn't be run frequently and they are
Most of our code which calls methods conditionally use `If`'s for it and as we know they block such methods and can cause conflicts which not exist in reality.

Of course we can do that manually, but I don't like to analyse few hundreds(?) methods by hand. We can also add synchronous process, but such test case have to be written first and methods to be analysed have to be chosen manually (this don't scale easily over many possible core configurations).

tilk, 2023-07-03:
> Of course we can do that manually, but I don't like to analyse few hundreds(?) methods by hand. 

You wrote that:

> So please add a script which as an input will take a name of method in a module (...)

So that's just as manual as looking in gtkwave, and now it looks like you want something that gathers and makes a readable visualization from a lot of data.

Also, your description, together with the "infrastructure" label, seemed to imply you want something to run in CI, not an analysis tool. Could you write a better description?

lekcyjna123, 2024-03-24:
Solved by #547 and #596

closed 2024-03-24 by lekcyjna123