kuznia-rdzeni/coreblocks #412: Using AI to write unit tests

lekcyjna123, 2023-07-01:
Today AI is very hot topic, so maybe we can see if it will be usable in our project (and if not we can laugh of it). I see that for students, teachers or open source projects Github Copilot is free, so maybe we can try to use it to write unit tests?
https://docs.github.com/en/copilot/overview-of-github-copilot/about-github-copilot-for-individuals#about-billing-for-github-copilot

lekcyjna123, 2023-07-01:
Some open source models which I have found:
https://github.com/salesforce/CodeT5
https://github.com/VHellendoorn/Code-LMs

tilk, 2023-07-03:
You aren't serious, right?

lekcyjna123, 2023-07-03:
I have no hopes that AI will write whole tests, but I am interested if we can use it as better auto-completion engine, which can generate some boiler plate. I am also interested what are the current limitations of todays models, what can they do and what not and I thing that our project is a good playground for such experiments.

tilk, 2023-07-03:
Sure, have fun, but this is not an AI project, I wouldn't want it to get sidetracked.

closed 2023-07-07 by lekcyjna123