kuznia-rdzeni/coreblocks #416: Add support for one caller methods in typing

lekcyjna123, 2023-07-02:
Since we can make methods ready based on input data in one caller methods, such usage is being added continuously in new places to make some _optimisations_. As for now we don't have any support for such methods in typing system and they are treated as normals methods. Only docstrings can said that if method will be connected to two or more callers an combinational loop will occur. So to prevent such kind of errors I would like to ask for introducing OneCallerMethod type to our typing system.

tilk, 2023-07-03:
Single-caller should be checked at the transaction manager level, not the type level. Also, single-caller should probably be used sparingly.

lekcyjna123, 2023-07-03:
> Also, single-caller should probably be used sparingly.

Single caller can be easily transformed to multi-caller at cost of 1 cycle latency, so I would rather not introduce any limitations for single-caller usage, as long as place were such method was used make sense. It is particularly useful for methods with `condition`, to don't block methods used in all branches if condition depends on input data.

https://github.com/kuznia-rdzeni/coreblocks/blob/8f3fc5ecf00a4d2b420640e0c60734e27eb0039c/coreblocks/transactions/lib.py#L1297-L1305

PR #425 (2023-07-05): Single caller method check
For @lekcyjna123. For some weird, but kinda sensible use cases, it makes sense to have a method which can be called at most once. In such a case, it's better to check that assumption to avoid surprises later.

Fixes #416.
closed 2023-07-05 by tilk