kuznia-rdzeni/coreblocks #444: Analyse synthesis cost of FuDecoder

lekcyjna123, 2023-07-13:
On the beginning of the project we did an assumption that decoding in FU will consist of two steps. First we decode `exec_fn` to internal one-hot representation and next we do second switch to execute proper operations.

This way of working have been grounded when we didn't have tools to compare efficiency in terms of Fmax and LUT usage. Now we have some, so it will be nice to check if our decision was right or if we should optimise our decoders.

ToDo:
- Take any FU (e.g. ALU) synthesise it without rest of the core and check Fmax and LUT usage
- rewrite chosen FU so that it use only one switch instead of two (don't use intermediate representation with IntFlag) and check Fmax and LUT usage

tilk, 2023-07-13:
This would be kind of weird, as:

* In silicon, binary multiplexers are basically just one-hot multiplexers with binary decoders.
* In FPGA, the two stages might just get glued together in LUTs anyway.

And the intermediate `IntFlag` serves an useful purpose, standardizing decoders, implementations and testing. The change you propose might complicate stuff. I don't think it is worth it to work on this. Convince me.

I have a different question on my mind: is it worth it to move FU decoding to some earlier stage?

lekcyjna123, 2023-07-13:
Both of this points assume that synthesis tools are able to do proper optimisations, so glueing together both switches and observing that in reality there is only one binary multiplexer. But we use a lot of abstractions, and a higher level language, which has to be translated first to RTLIL, so in my opinion we can not be sure that after all intermediate steps optimiser will be able to do its job properly.

> I have a different question on my mind: is it worth it to move FU decoding to some earlier stage?

Yest this is also an interesting point.