kuznia-rdzeni/coreblocks #446: Project cleanup

tilk, 2023-07-14:
There is a need to move stuff around to make things more intuitive for newcomers (and others). Some problems in directory organization are:

* The `coreblocks/params` directory is a mess. It was supposed to contain things related to parametrization (at the beginning, mostly `GenParams`). Now it also contains:
  * `dependencies` - the dependency manager,
  * `keys` - the dependency keys (which we agreed to have in one place),
  * `isa` and `instr` - RISC-V architecture related things,
  * `optypes` - our internal encoding of instructions, which is entangled with ISA, decoding and FUs.
* The `structs_common` directory somehow contains a CSR unit. The directory was meant to contain implementation of data structures, i.e. things that are shared between components.
* The `coreblocks/transactions/lib` is one big ball of stuff, it should finally be split.
* The `stages` directory is weird, it contains:
  * The announcement module, in a file which is somehow named `backend`,
  * The retirement stage,
  * Abstractions related to functional units/blocks.

There are potentially other issues I can't remember right now, please write them down in comments.

lekcyjna123, 2023-07-14:
Another issues:
- move whole `coreblocks/transactions` to `transactron` directory so that it will be easier to split it in future to separate repository
- cleanup `utils`
  - fifos use transactional interface so they should be part of `transactions/lib`
  - some modules can be send to amaranth upstream
  - `_typing` is a ball of stuff for types, on the begining it was ment to consist only internal types, but now it has types representing amaranth concepts, which we use in `transactron` interfaces
- `transactions/core` also is huge, it should be splitted (e.g. to definitions used by developers such as `Method`, `Transaction` and transactron internal stuff such as `methodMap`)
- I would suggest to split syntax sugar to separate files so that it will be more clear that it isn't strictly needed to use it
- `test/common.py` need splitting
  - useful functions should go to one place
  - classes used on test case creation (SimpleTestCircuit, TestCaseWithSimulator)
  - testbenchIO
  - syntax sugar
- `ci` and `scripts` directories should be in my opinion merged
- I would suggest splitting `coreblocks/scheduler/scheduler.py` to smaller files
- `_instructions_by_optype` should be moved from `coreblocks/frontend/decoder.py` to separate file (after adding part of vector extension it take more than 270 lines to define all instructions)

lekcyjna123, 2023-07-14:
If this is possible I would ask to wait with implementation of this cleanup till interrupts and vectors will be merged to master. I plan to set vectors as ready for review in Sunday/Monday (I am now integrating and testing all vector parts).

tilk, 2023-07-16:
> If this is possible I would ask to wait with implementation of this cleanup till interrupts and vectors will be merged to master.

I don't have any estimate on how long interrupts will be in development (maybe @Kristopher38 needs some help there?), but I agree that waiting for them is okay. I would prefer the cleanup to happen before October, though. As for vectors, I doubt they will be mergable in short timeframe. This functionality will probably require multiple PRs and piecewise review.

Kristopher38, 2023-07-19:
For the love of me I can't reproduce amaranth generating `assign ""` on my local machine. I've tried with both python3.10 and python3.11 without success. It's really weird - I'm using the same amaranth commit and PYTHONHASHSEED as is used on the CI. How can I download the docker image used in the CI? Doing `sudo docker pull ghcr.io/kuznia-rdzeni/verilator:v5.008` says I'm unauthorized.

As for interrupts development, iirc @piotro888 said something about doing further work? I'm taking a (well-deserved after last semester's crunch) break at least until September, though maybe I'll feel inspired to contribute something before that ;)

tilk, 2023-07-20:
> How can I download the docker image used in the CI?

The package visibility was set to "internal", I've changed it now to "public". If this won't help, you probably need to authenticate.

lekcyjna123, 2023-10-24:
Another point to do is to cleanup our documentation webpage.

lekcyjna123, 2024-01-08:
Move transactron part of tests to `transactron` directory.

lekcyjna123, 2024-03-13:
One more proposition: https://github.com/kuznia-rdzeni/coreblocks/pull/617#pullrequestreview-1932978170

tilk, 2024-12-05:
This is basically done.