kuznia-rdzeni/coreblocks #448: Run tests in loop

lekcyjna123, 2023-07-17:
Currently each unit test is parameterized using two values important from running test case: seed and test_number. First set seed for random generator and second said us how many iteration we should do for each seed. When we want to increase coverage we have two ways:
- test with different seed's
- test with more iterations
Both ways are orthogonal, the first is better if we want to find shorter sequence of iterations which gave us the same result as some other longer tests, and the second is better when we want to check how our core behaves in long run.

As for now if someone want to change seed he has to either edit manually test case or wrap it with `parameterized_class` to test more seed's in one `run_tests.py` command. It would be nice to have a possibility to specify in `run_tests.py` number of loops to execute. Then this script should automatically set for each unit test run a different seed and test with it.

If implemented correctly, this will have an another pros, that we won't have to remember about setting seed manually and this will remove some boilerplate code from tests.

tilk, 2023-07-18:
> each unit test is parameterized using two values important from running test case: seed and test_number.

You mean, unit tests which use random test generation. We have a number of completely deterministic unit tests.

I can understand the need to (sometimes) run tests differently, to make sure some obscure bug wasn't missed. On the other hand, the main role of unit tests is to help catch bugs when doing changes to code; default test parameters should be set by the test developers so that the probability of missing a bug is low.

Managing RNG seed at the test runner level looks like something definitely worth having.

Kristopher38, 2023-07-19:
This is partly how ibex does it. They run nightly tests with various seeds and see how many of them pass https://ibex.reports.lowrisc.org/opentitan/latest/report.html

On that note there's also a great presentation that explains a bit more how they're testing their core - there's a lot of approaches that we should implement sooner or later https://www.youtube.com/watch?v=EYyi8ZhEQxA&list=PL85jopFZCnbMlDT5WOagUjQnk1ZSgjuRa&index=13