kuznia-rdzeni/coreblocks #451: Use riscv-dv for randomized core testing

tilk, 2023-07-20:
[RISCV-DV](https://github.com/chipsalliance/riscv-dv) is an instruction generator, which can be used for automatic testing of RISC-V cores. For execution, our existing Verilator+cocotb solution can probably be adapted. To verify execution, results should be compared to a model, for example the [Sail model](https://github.com/riscv/sail-riscv).

lekcyjna123, 2023-07-20:
Another possibility is to use [Spike](https://github.com/riscv-software-src/riscv-isa-sim), which supports a huge number of extensions. Additionally maybe it can be worth to check if [Riscof](https://riscof.readthedocs.io/en/stable/) can be used.

awariac, 2026-08-16:
@kiryk will be taking a look