kuznia-rdzeni/coreblocks #494: Implement physical memory attributes (PMA) for memory-mapped IO (MMIO)

tilk, 2023-10-30:
Physical memory attributes (PMA), as described in the second volume of the RISC-V ISA manual, serve to describe characteristics of the physical memory. They can describe properties like cachability, misaligned access support, memory ordering and idempotency. For now, most of these things are not needed for our purposes, but *idempotency* is extremely important for memory performance and correctness.

Normal memory is idempotent, which means that:

* Multiple identical writes to the same address are equivalent to a single write.
* Multiple reads from the same address, without writes in the middle, will return the same value.

These properties can be used to optimize memory access. Memory mapped I/O is, in contrast, not idempotent:

* Each write to an address might cause a different effect (e.g. different character sent over UART).
* Each read from an address might get a different result, because of changing state of a peripheral (e.g. counter advancing) or initiating an action (e.g. character received over UART).

Memory ordering is also very important for working MMIO:

* Memory-mapped I/O in general requires strong ordering: reads and writes cannot be reordered in any way.
* Normal memory can have relaxed ordering: reads and writes can be reordered as long as they fit the used memory model (RVWMO or TSO).

The RISC-V spec doesn't suggest a physical implementation of PMAs, so I suggest a simple one: a read-only one, which is created according to the core configuration.

piotro888, 2023-10-30:
Note/hint: for current `DummyLSU` only condition `is_load` in line https://github.com/kuznia-rdzeni/coreblocks/blob/e484fde04b76d1c83302cc6f4bedf8890b2c3a40/coreblocks/lsu/dummyLsu.py#L132C18-L132C18 needs to be 'disabled' for MMIO reads to delay its execution to 'safe' moment (when writes are executed by `precommit`)

kindlermikolaj, 2023-11-04:
> Note/hint: for current `DummyLSU` only condition `is_load` in line [...]

As far as I understand how DummyLSU works, such scenario when there is MMIO write yet to be executed  and DummyLSU processes MMIO read is impossible, since it would be serialized and executed already?

tilk, 2023-11-05:
> As far as I understand how DummyLSU works, such scenario when there is MMIO write yet to be executed and DummyLSU processes MMIO read is impossible, since it would be serialized and executed already?

You seem to misunderstand. What @piotro888 tried to tell you about is that, in DummyLSU, MMIO writes are no problem at all as all writes are processed at the time they are just to be committed. The problem is exclusively with MMIO reads. The DummyLSU currently starts reads as soon as possible after the instruction is loaded into the core. It might happen that e.g. a previous instruction gets an exception, and the load is never committed. This is incorrect for MMIO, because MMIO reads may have side effects: a side effect happens for an instruction which, logically, never executed.

kindlermikolaj, 2023-11-05:
Ok, now I understand.

closed 2023-12-12 by tilk