kuznia-rdzeni/coreblocks #556: Fix retirement test

tilk, 2024-01-06:
It was missed in review that test_rand in retirement tests has a yield outside of test processes.

piotro888, 2024-01-06:
I found it and unfortunately it is due to `# TODO: mocking really seems to dislike nonexclusive methods for some reason`. In new code `rob_peek` is called from two transactions and simulation hangs up.

I also fixed other bugs, but the test wont run without this one solved.

piotro888, 2024-01-06:
Even manual mocking like this:
```python
 def peek_process():
            yield from retc.mock_rob_peek.enable()
            yield Passive()
            while self.submit_q:
                yield from retc.mock_rob_peek.call_init(self.submit_q[0])
                yield
```
fails when nonexclusive method is called multiple times in dut

piotro888, 2024-01-07:
> `# TODO: mocking really seems to dislike nonexclusive methods for some reason`.

Fixed in #558

PR #562 (2024-01-11): Retirement test fix
Closes #556 
Adds missing mocks, fixes `nonexclusive`and resolves todo with retire/peek functionality duplication
(test was not running at all because of yield that lekcyjna123 found)
closed 2024-01-11 by tilk