kuznia-rdzeni/coreblocks #561: Implement machine-mode CSRs

tilk, 2024-01-08:
The CSRs in question are:

* `misa`
* `mvendorid`
* `marchid`
* `mimpid`
* `mhartid`
* `mstatus` and `mstatush`
* `mtvec`
* `mip` and `mie`
* `mcycle` and `minstret`
* `mscratch`
* `mepc`
* `mcause`
* `mtval`
* `mconfigptr`

piotro888, 2024-01-11:
@tilk allocation for `marchid`. Are names and email good? Approved?
```diff
diff --git a/marchid.md b/marchid.md
index 7f85b26..f4a0be0 100644
--- a/marchid.md
+++ b/marchid.md
@@ -59,3 +59,4 @@ WIV64         | Jesús Sanz del Rey              | [Jesús Sanz del Rey](mailto:
 RV6           | Nikola Lukić                    | [Nikola Lukić](mailto:lukicn@protonmail.com)                | 39               | https://github.com/kiclu/rv6
 ApogeoRV      | Gabriele Tripi                  | [Gabriele Tripi](mailto:tripi.gabriele2002@gmail.com)       | 40               | https://github.com/GabbedT/ApogeoRV
 MicroRV32      | AGRA, Group of Computer Architecture, University of Bremen | [RISC-V @ AGRA](mailto:riscv@informatik.uni-bremen.de)       | 41               | https://github.com/agra-uni-bremen/microrv32
+Coreblocks    | Kuźnia Rdzeni, University of Wrocław | [Coreblocks Team](mailto:coreblocks@cs.uni.wroc.pl)    | 42               | https://github.com/kuznia-rdzeni/coreblocks
```

tilk, 2024-01-11:
@piotro888 Approved.

PR #743 (2024-11-12): Implement `misa` CSR
Closes #561
Complete machine mode CSRs :tada:
closed 2024-11-12 by tilk