kuznia-rdzeni/coreblocks #579: Timing-out unit tests of the core

xThaid, 2024-02-01:
I found an interesting bug, but I am not sure how to correctly handle it. I am also making this issue, because it remains documented and maybe will be helpful to someone else.

## The issue
The unit tests with my changes from PR #572 started to time-out in the CI. When I run it locally, it finishes successfully, but it takes longer (around 6 min longer). I found out that the culprits are the core tests `test.test_core`. It seems that functionality-wise metrics didn't break anything. 

Here is the interesting part. I measured the time of running the tests:
With the metrics disabled:
```
time scripts/run_tests.py test.test_core
Ran 12 tests in 198.386s

OK

real	3m18.919s
user	3m13.351s
sys	0m5.352s

```
With the metrics enabled:
```
$ time scripts/run_tests.py test.test_core
Ran 12 tests in 440.994s

OK

real	7m21.566s
user	3m38.665s
sys	3m42.589s
```
The time increase is almost only caused by some in the kernel? Indeed there is something fishy going on. 

I ran `strace` to see what is happening and the reason of that unreasonably high time spent in the kernel became obvious:
```
# strace scripts/run_tests.py test.test_core | grep "mmap" | wc -l
20370324
```
The Python runtime repeatedly calls:
```
mmap(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f5c05057000
munmap(0x7f5c05057000, 16384)           = 0
``` 
It doesn't seem that this is (directly) caused by a bug in Amaranth. When the syscall is called, Amaranth sits in `amaranth/sim/pysim.py, line 213`, which is just a hot path of the simulation.

It seems that someone had similar issue in the past: https://stackoverflow.com/questions/36548518/why-is-code-using-intermediate-variables-faster-than-code-without. Tldr: the python may be preemptively releasing memory to the kernel and then right after taking it back, but that is just my hypothesis.

To verify it, I added another histogram instance in the core to increase memory footprint of the simulation. I hoped that maybe in that case Python won't be trashing memory. And it actually worked - with another histogram to simulate, the run time of the tests was back to the normal values.

## Solutions
a) just add more metrics in my PR, so it doesn't timeout anymore
b) increase timeouts in the CI
c) dig it further (not a fan of this one)

## Final remarks
I was thinking if we need the core tests (as unit tests) at all. They are the slowest parts of our testing pipeline. They could be migrated to our regression test framework, so by default they would run with cocotb but if debugability is needed, pysim backend would be always available. Some of the core tests we could remove altogether (`TestCoreSimple`, `TestCoreRandomized`), as they don't seem to be effectively testing the core and just gives us another different type of tests to maintain.

tilk, 2024-02-03:
That's quite a weird situation. I think just working around it would be enough. Preferably option (a), if it works reliably.

I agree that some of the tests are vestiges from the time when we couldn't do better. It would probably for the best to remove them.

Kristopher38, 2024-02-04:
Agreed, I'd remove all core tests on pysim backend save for the assembly tests which should be replaced with regression tests. I'd leave the code for handling them in the codebase though, as they're sometimes useful for testing new core features.
One exception is the interrupt tests, they should stay in place until we implement full interrupts support and can move on to testing them outside of our python setup.

closed 2024-03-05 by xThaid