kuznia-rdzeni/coreblocks #592: Fix B extension

piotro888, 2024-02-29:
Regression tests are failing for the following B extension instructions:

- [ ] `clz`
- [ ] `cpop`
- [ ] `ctz` 
- [ ] `orcb`
- [ ] `rev8`
- [ ] `rol`
- [ ] `ror`
- [ ]  `rori`
- [ ]  `sext.b`
- [ ]  `sext.h`
- [ ]  `zext.h`

https://productionresultssa1.blob.core.windows.net/actions-results/b6033def-fe7d-49ca-94fe-90b456585539/workflow-job-run-3ffad427-3fce-534e-df7b-6607fc34e12a/logs/job/job-logs.txt?rsct=text%2Fplain&se=2024-02-29T10%3A08%3A14Z&sig=FDIia%2FpzwXQOAy8YwzaIPK7dnOPzMZzJuTnezCOdE5c%3D&sp=r&spr=https&sr=b&st=2024-02-29T09%3A58%3A09Z&sv=2021-12-02

xThaid, 2024-02-29:
The link does not work for me.

piotro888, 2024-02-29:
correct link https://github.com/piotro888/coreblocks/actions/runs/8093851411/job/22117470063 , "view raw logs" may be needed if compare log is truncated

piotro888, 2024-02-29:
I found a problem with instruction decoding, working on fix

closed 2024-02-29 by tilk