kuznia-rdzeni/coreblocks #603: Running regression tests with pysim backend

tilk, 2024-03-06:
I'm trying to run regression tests in my working directory using this command:

```
scripts/run_tests.py --all --verbose -b pysim regression
```

This is the error I'm getting:

```
fixturefunc = <function verilate_model at 0x7fc811c0bba0>, request = <SubRequest 'verilate_model' for <Function test_entrypoint[rv32uc-rvc-pysim-False-True]>>
kwargs = {'request': <SubRequest 'verilate_model' for <Function test_entrypoint[rv32uc-rvc-pysim-False-True]>>, 'worker_id': 'gw0'}

    def call_fixture_func(
        fixturefunc: "_FixtureFunc[FixtureValue]", request: FixtureRequest, kwargs
    ) -> FixtureValue:
        if is_generator(fixturefunc):
            fixturefunc = cast(
                Callable[..., Generator[FixtureValue, None, None]], fixturefunc
            )
            generator = fixturefunc(**kwargs)
            try:
                fixture_result = next(generator)
            except StopIteration:
>               raise ValueError(f"{request.fixturename} did not yield a value") from None
E               ValueError: verilate_model did not yield a value

.venv/lib/python3.11/site-packages/_pytest/fixtures.py:918: ValueError
```

This feels wrong. Why does pysim even care about `verilate_model`? I don't know enough about `pytest` to debug this quickly.

lekcyjna123, 2024-03-06:
I will take a look. I am able to reproduce it locally...

lekcyjna123, 2024-03-06:
> Why does pysim even care about `verilate_model`

Pysim doesn't care about the `verilate_model`. It is a common session fixture which is early ended without any actions when backend is `pysim`. The bug was in this early ending. There was lacking `yield` which should separate session set up from session teardown.

PR #604 (2024-03-06): Fix the pysim regression
Here is quick fix after #554. We don't have tests of all option combinations of our `run_tests.py` script so when the regression with pysim stopped to work after one of the refactor it wasn't catched on CI. CI will be extended in separate PR.

Resolves #603.
closed 2024-03-06 by tilk