kuznia-rdzeni/coreblocks #610: Single-core compilation of verilated core

tilk, 2024-03-08:
Even though the new testing system allows the use of multiple cores, compilation of the C++ files generated by Verilator is single-core, which makes the running time of regression tests somewhat longer than it could be.

xThaid, 2024-03-09:
As far as I understand, it is not even the problem with compilation, but cocotb tests are just run sequentially. That is, we call cocotb for every regression test and the overhead of calling cocotb is non-neglible (even assuming that c++ files are already compiled). Previously, we called cocotb once and asked it to run all tests.

tilk, 2024-03-10:
There is definitely a problem with compilation. When I run regression tests locally, most of the time is spent in compilation (a single GCC process at a time is using the CPU). After that, the tests run comfortably.

PR #623 (2024-03-21): Allow parallel compilation of the verilated model
Closes #610.
closed 2024-03-21 by tilk