kuznia-rdzeni/coreblocks #628: Implement Zicond

tilk, 2024-03-25:
The Zicond extension extends the architecture with conditional instructions, which allow to create code with less branches. It's probably the best to implement this in ALU.

Hazardu, 2024-11-02:
When trying to implement Zicond, I came across enums of different shapes named `Funct3` and `Funct7` in [isa_consts.py](https://github.com/kuznia-rdzeni/coreblocks/blob/master/coreblocks/arch/isa_consts.py#L49-L72) but I can't find any descriptions what they are supposted to describe exactly and what their values mean. 

I see that they are used when defining the encodings of operation types in [instr_description.py](https://github.com/kuznia-rdzeni/coreblocks/blob/master/coreblocks/frontend/decoder/instr_description.py#L47)
```
Encoding(Opcode.OP, Funct3.ADD, Funct7.ADD),
``` 
as well as used differently in [alu.py](https://github.com/kuznia-rdzeni/coreblocks/blob/master/coreblocks/func_blocks/fu/alu.py#L66) to describe instructions
```
(self.Fn.ADD, OpType.ARITHMETIC, Funct3.ADD, Funct7.ADD),
```

but I dont understand what their values represent (like previousely mentioned  `Funct3.ADD` and `Funct7.ADD` equal  `0`) 
Could you please explain?

piotro888, 2024-11-03:
[image]
@Hazardu `Opcode/Funct*`  are fields from raw RISC-V instruction encoding.

`Optype` is our internal representation of instruction groups (as replacement of `Opcode`), used also as an only factor for routing instructions to Functional Units.  `Optype` is resolved in Decode stage.

Functional units have their own small internal decoders that convert `Optype` (from instruction_description.py) and `Funct3`, `Funct7`  (from raw instruction; Funct7 is not always present or needed [if Optype+Funct3 are unique]) to specific unique FU operation (to `self.Fn`)

Zicond should have separate Optype as it is separate extension.

closed 2024-11-26 by tilk