kuznia-rdzeni/coreblocks #629: Add support for array layouts in the method layout

xThaid, 2024-03-26:
We could leverage Amaranth's new aggregate data structure, `ArrayLayout`, to add support for method layouts that takes an array (of a fixed size) of arguments.

The syntax could be like:
```
layout = [
                ("instr", self.gen_params.isa.ilen, fetch_width),
                ("foobar", 1),
]
```
That is, if a layout field is a tuple of 3 elements, we treat it as an array where the last element of the tuple is the size of an array.

tilk, 2024-03-26:
Currently, an `ArrayLayout` should be possible to be used by just writing it in place of the bit width. I haven't tested this though.

xThaid, 2024-03-27:
> Currently, an `ArrayLayout` should be possible to be used by just writing it in place of the bit width. I haven't tested this though.

The layout itself works, but `assign` and perhaps something else need to be updated accordingly.

tilk, 2024-03-27:
Indeed, only `StructLayout` is currently supported in `assign` - support for other layout types needs to be added there.

xThaid, 2024-04-03:
Solved by #640

closed 2024-04-03 by xThaid