kuznia-rdzeni/coreblocks #672: Red/green arrows in benchmark bot comments

tilk, 2024-04-19:
It would be nice if the arrows in the benchmark bot comments were colored, to signify changes for better or worse. This essentially requires two things:

* Knowing which metric is if-bigger-then-better, and which is if-smaller-then-better.
* Displaying the arrows. Markdown doesn't support font colors by itself, but it might be possible to use embedded SVGs or LaTeX snippets to do this.

ArturBieniek4, 2025-03-16:
The following LaTeX snippet:
`$${\color{red}▲}$$`
Produces $${\color{red}▲}$$.
Unfortunately I have no idea which metric is if-bigger-then-better and which is if-smaller-then-better.
Is it OK to create a dict indicating so directly in https://github.com/kuznia-rdzeni/coreblocks/blob/master/ci/print_benchmark_summary.py or should this information be somewhere else?

tilk, 2025-03-17:
Instructions per cycle (IPC) are if-bigger-then-better. So is maximum synthesis frequency (Fmax). LCs are if-smaller-then-better (less FPGA resources used).

> Is it OK to create a dict indicating so directly in `print_benchmark_summary.py` or should this information be somewhere else?

Hardcoding it is probably not a great idea. The set of benchmark results might change in the future, and this would require remembering to also change the dict. Also, different benchmark result files have different information: for example, the performance benchmarks (currently) emit only IPC values for a given set of benchmarks; the set of benchmarks might change, and other kinds of metrics other than IPC might get added (like amount of cycles, number of mispredictions, etc.). Synthesis benchmarks have Fmax and resource usage collected; but we might use other FPGA toolchains later, or gather other synthesis metrics.

The benchmarking mechanism could probably benefit from a refactor, for example decoupling the output format of benchmark scripts from the input format for the graph drawing action we use currently.