kuznia-rdzeni/coreblocks #772: MethodStruct is not defined error when running tests with editable transactron

Hazardu, 2024-12-08:
I'm trying to run pytest in coreblocks with an editable transactron checked out from [this commit](https://github.com/kuznia-rdzeni/transactron/commit/6520af4774f198c998a28ec0eb9ad198e04b4508). 
Pytest is throwing me 32 erros that MethodStruct is not defined.

```
ERROR test/backend/test_annoucement.py - NameError: name 'MethodStruct' is not defined
ERROR test/backend/test_retirement.py - NameError: name 'MethodStruct' is not defined
ERROR test/cache/test_icache.py - NameError: name 'MethodStruct' is not defined
ERROR test/core_structs/test_rat.py - NameError: name 'MethodStruct' is not defined
ERROR test/core_structs/test_reorder_buffer.py - NameError: name 'MethodStruct' is not defined
ERROR test/frontend/test_decode_stage.py - NameError: name 'MethodStruct' is not defined
ERROR test/frontend/test_fetch.py - NameError: name 'MethodStruct' is not defined
ERROR test/frontend/test_instr_decoder.py - NameError: name 'MethodStruct' is not defined
ERROR test/frontend/test_rvc.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/csr/test_csr.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/common/test_rs.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_alu.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_div_unit.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_exception_unit.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_fu_decoder.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_jb_unit.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_mul_unit.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_pipelined_mul_unit.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_shift_unit.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_unsigned_mul_unit.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_zbc.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/fu/test_zbs.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/lsu/test_dummylsu.py - NameError: name 'MethodStruct' is not defined
ERROR test/func_blocks/lsu/test_pma.py - NameError: name 'MethodStruct' is not defined
ERROR test/params/test_configurations.py - NameError: name 'MethodStruct' is not defined
ERROR test/priv/traps/test_exception.py - NameError: name 'MethodStruct' is not defined
ERROR test/regression/cocotb/test_entrypoint.py - NameError: name 'MethodStruct' is not defined
ERROR test/regression/test_regression.py - NameError: name 'MethodStruct' is not defined
ERROR test/scheduler/test_rs_selection.py - NameError: name 'MethodStruct' is not defined
ERROR test/scheduler/test_scheduler.py - NameError: name 'MethodStruct' is not defined
ERROR test/scheduler/test_wakeup_select.py - NameError: name 'MethodStruct' is not defined
```

there doesnt seem to be missing imports anywhere, the only clue is that MethodStruct seems to be a forward reference in [`transactron.utils._typing`](https://github.com/kuznia-rdzeni/transactron/blob/6520af4774f198c998a28ec0eb9ad198e04b4508/transactron/utils/_typing.py#L51)
```
MethodStruct: TypeAlias = "View[StructLayout]"
```

tilk, 2024-12-08:
The commit you're referencing removed some spurious exports. You just need to add appropriate imports where needed.

tilk, 2024-12-08:
And also, because this is unrelated to whatever you're doing, this should be in a separate PR which bumps Transactron version.

closed 2024-12-09 by piotro888